SynthstromAudible/DelugeFirmware #12: Create second upstream USB midi port

m-m-adams, 2023-06-06:
A second USB midi upstream port should be enabled to separate MPE zones from normal midi channels. I am happy to work on this

litui, 2023-06-07:
Cool idea!

m-m-adams, 2023-06-09:
Making an initial to do list:

1. Change the usb midi device configuration in r_usb_pmidi_descriptor to show 2 extra usb endpoints (on the same device), ref - https://www.usb.org/sites/default/files/midi10.pdf
    1a. Currently this is using bulk endpoints, should investigate if there's a downside to using more bulk endpoints or if they should be changed to a different subtype
    Source - r_usb_pmidi_descriptor.c, r_usb_basic_if.h

2. Modify the USB Midi driver to distinguish multiple upstream ports
    source - r_usb_hlibusbip.c

3. Modify the midi device setup to distinguish multiple upstream ports
    source - MIDIDeviceManager.cpp

closed 2023-06-20 by m-m-adams