SynthstromAudible/DelugeFirmware #7: Cross-platform Build Scripts (Windows, Mac, OSX)

litui, 2023-06-06:
This is in the works! Stay tuned!

The scripts will be in a combination of Bash and Python using SCons as the make system, borrowing liberally from the excellent cross-platform firmware build scripting at https://github.com/flipperdevices/flipperzero-firmware (also under GNU GPL).

litui, 2023-06-12:
I've pivoted a bit in my direction here. While the auto-downloader and environment changing code will still be used, I'm reworking the SCons approach considerably to take into account settings in the .cproject XML config files. This will allow e2 to "run the show" as far as build settings go.

chrisbc, 2023-06-14:
@litui do you think this approach can eventually support cloud builds e.g. a pipeline using Github actions?

litui, 2023-06-19:
@chrisbc Sorry for the delayed response! I've been hunkering down to work on this. Absolutely it will, and that was one of my intentions with it. Check out PR #69 for the first release of it. =)

closed 2023-06-19 by litui