SynthstromAudible/DelugeFirmware #19: Implement multiple note layouts in keyboard View

MilesParker, 2023-06-07:
The current keyboard view (i.e. "playing" Deluge notes ) supports a +5 semitone (fourth) row offset / tuning. This task is to allow users to set different layouts for display. Reasonable options run from +3 to +8 (octave).

UX TBD, but one possibility is shift + piano_keys + <> (| ^v). (Alternatives?)

(We could also consider other irregular options such as guitar, with the differing G-B offset, or a whole tone layout. As that looks like a more complex change it probably makes more sense as a separate task.)

For a similar implementation, see "Row Offset" in Linnstrument options  #below:
https://www.rogerlinndesign.com/support/linnstrument-support-panel-settings

MilesParker, 2023-06-07:
(Enhancement, can't seem to add label..)
Willing to take a go at this one but won't be able to make progress until my Deluge gets back from upgrade. But feel free to tackle it.

sichtbeton, 2023-06-07:
> The current keyboard view (i.e. "playing" Deluge notes ) supports a +5 semitone (fourth) row offset / tuning. This task is to allow users to set different layouts for display. Reasonable options run [...]

Would it be possible to have a note layout for kits so you can play all kit items? maybe even include something like a 16 levels approach for certain kit items?

MilesParker, 2023-06-07:
That's a really neat idea, kind of a pads mode. Though not really in scope for this particular issue.

bfredl, 2023-06-11:
> Willing to take a go at this one but won't be able to make progress until my Deluge gets back from upgrade. But if anyone else is inspired, go for it. :)

I took stab at this (at least the variable fixed offset part) as I really need it with the ongoing microtuning work: https://github.com/SynthstromAudible/DelugeFirmware/pull/46 .

PaulFreund, 2023-07-01:
@MilesParker Do you still want to work on this? I'm planning to start work on a PR with a multi layout (piano, isomorphic, fixed), configurable (rows, columns, in-key only, chords) keyboard that should implement #19 and #55 extending on #46 and don't want to collide with anybody

MilesParker, 2023-07-02:
Hi @PaulFreund -- thanks for checking in ... I don't actually have my Deluge now anyway as it is off being upgraded, so please go for it! LMK if you want to bounce any ideas around. :)

PaulFreund, 2023-07-28:
Hey wanted to check back, keyboard API is implemented. Please try it out, the UX is shown in the linked Video (changing row offset also works in Isomorphic).

closed 2023-07-30 by m-m-adams