SynthstromAudible/DelugeFirmware #27: Establish Contributing Guidelines

BenRGH, 2023-06-08:
As per the standard for community open source projects, it makes sense that there should be a Contributing Guidelines defined in the repository, à la defined in the github docs [here](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors). 

This would help enforce a standard for development and prevent wasted time correcting PR's with conflicting approaches or formats etc.

litui, 2023-06-09:
'm happy to make a starting point for this and submit a PR.

litui, 2023-06-09:
If folks could review the development branch to the right I'd appreciate it ------->

Please add your additions and thoughts in the PR thread and I'll try to incorporate them.

closed 2023-06-12 by litui