SynthstromAudible/DelugeFirmware #28: Ableton-like clip launching

stosem, 2023-06-08:
It would be nice to have a new "Settings" option that allows us to switch the classic Deluge clip lunch layout to something similar to the Ableton/Bitwig/Logic view.
With columns of instruments and rows of patterns.
We can launch clips by clicking on them, or launch an entire "scene" consisting of patterns in a row.
The record function allows to store the clip launching sequence in the arranger mode (similar to the Push functions).

designerfuzzi, 2023-06-14:
Would make also sense to allow the 8x16 button matrix and 2x8 Right side buttons to be used as Midi controller (aka launchpad) with a specialised view, maybe even allowing different views accessible via Menu (somehow). If we could manage to develop a simple midi protocol one could write MIDI controller scripts targeting Clip Launch of a DAW and similar.
From there it would be a total logic step to just allow such for the Deluges own content/clips.

weavermedia, 2023-07-19:
There's discussion about this feature in an RFC here: https://github.com/SynthstromAudible/DelugeFirmware/discussions/150

PaulFreund, 2023-07-28:
Work is now tracked in #251 :)

closed 2023-07-30 by m-m-adams