SynthstromAudible/DelugeFirmware #41: 7SEG Emulation Mode

bobtwinkles, 2023-06-09:
@bfredl mentioned this on the discord, https://discord.com/channels/608916579421257728/1107026299945496577/1116778630656299098.

> maybe this already has been discussed but one thing which I'd like to see: A "virtual 7seg" build mode. Build the entire UI/menu system with HAVE_OLED set to false, but add a separate flag to still build the low-level oled driver. then write an oled rendering routine which emulates the 7seg display.
> The main usecase is of course debugging/testing ui code, if you have an OLED deluge you can debug how your changes will behave on a 7seg deluge

In the general spirit of Synthstrom's approach to supporting the 7seg, we probably want to make sure any features (enabled by default only?) that get merged to `community` support the 7seg well.

--- 

Technical sketch:
- Use the existing `HAVE_OLED` to mean that we're using the OLED UI, to minimize churn in the upper/UI layers
- Introduce a new define, `DELUGE_DISPLAY` (bikeshed name) similar to `DELUGE_MODEL` which is used by the lower layers to switch between physical OLED and 7SEG output.

ok-reza, 2023-06-13:
I've uploaded assets and some guides based on the dimensions of the OLED's screen.

Currently, I've created two different font styles, Regular and Bold. Regular resembles the weight and spacing of the 7-SEG Deluge the most, whereas Bold takes liberties for better readability. (Excuse any anti-aliasing GitHub may introduce to the following images)
[image]
[image]

However, the Deluge's 7-SEG display is also slanted, I could make versions like that for both Regular and Bold pretty easily now, however I'm holding off until further progress is made with the current assets.  Readability will be a little more odd with them slanted anyway. Also there may be issues with incorporating the period (.) with slanted text due to it sharing space with the slanted text directly above it. Just depends on how things are implemented I suppose!

Once the fonts are properly implemented, the following guides will assist in formatting/arranging the two font styles properly on screen.
[image]

[image]

Also, while making these I wondered if displaying text via glyphs (how OLED seems to currently handle text) would be disadvantageous to displaying text by referencing which letters call for which segments to be 'lit'. There are a few formatting rules with the 7SEG display which may make it tricky to use with the OLED's text display code without some more finessing. That being said, I'm no programmer!

Happy to assist with design/art/assets where possible!

soymonitus, 2023-06-13:
Wow, love this idea

designerfuzzi, 2023-06-14:
@ok-reza in case you want to tinker with it, i recreated Deluges italic 7OLED font as `ttf` and `otf` a while ago and never really used it. [ttf on googledrive](https://drive.google.com/file/d/1mIYeQVh8dRzw-hmYbYVczIWltCpTWS9K/view?usp=sharing)
and [otf on googledrive](https://drive.google.com/file/d/1GvJEbCXla8DONm74kCjDjBTvFL4ButBB/view?usp=sharing)

annaluiseblume, 2023-07-08:
I would love to see some of this inspirational work being factored in:

https://youtu.be/RTB5XhjbgZA

m-m-adams, 2023-08-05:
Closing this as completed via https://bfredl.github.io/delugeclient/app.html and #192

closed 2023-08-05 by m-m-adams