SynthstromAudible/DelugeFirmware #51: Support for accidental notes

jantuitman, 2023-06-13:
(No PR done yet, I’ll first write my plan as an issue and I plan to start next week on it, but I am writing this to check if people have ideas or feedback)

Currently a clip can be edited with scale mode on (7 notes) or scale mode off (12 notes). The 7 notes views are far, far more easy to navigate since a full octave is always in range of the 8 vertical lines of the display. However, you cannot have notes outside the key once you turn this mode on.

This issue describes an enhancement that allows entering “out of scale” notes while scale mode is on, in a way that does not break the existing transpose functionality and key detection functionality.

## basic idea and data model

The basic idea is to add a pitch offset attribute to each note. This attribute starts on 0 when the note is created/placed. The existing scale detection and transpose logic can be unchanged, it will operate on the unaltered pitch only. 

## UI

How will the change be accessible in the UI:  it seems the most logical way to do this is pressing the note and turning a knob simultaneously. However, all encoder knobs already do something in combination with a pressed note. So I thought the most easy solution would be to have the main encoder, that also does note probability and 1:2, 1:3,2:3… and so on just display a second value after you press it to confirm the probability value.

so, note + press encoder : probability value shows (turn to change) , press again : accidental value shows (turn to change) , press again: back to probability.

## what will be displayed as accidental value.

If the offset value is 0 the string “UNC” (for LED) or text “Unchanged” will show. If an offset value is present then the pitch of the note after modifying it with the offset will be displayed. An alternative solution would be to display the offset itself as a signed number, but I think for most musical applications it will be more informative to see the pitch rather than the offset and still have to calculate the pitch in your head.
If you turn the encoder the offset value or displayed pitch will change. Whenever the offset will be 0 the text “UNC” will appear again.

# some open questions

   - it would be nice to color notes slightly different in a more distinguished way than they just being one semitone higher/lower once they have an non zero offset value assigned. I do not know how note coloring exactly works on the Deluge and I have no plan yet what a good informative way of displaying could be
   - if this idea is going to work I need to track down every usage of pitch in the sound engine because it may have to be modified to pitch + offset (with some max min clipping). Does this need to be precomputed and stored in the note or will it not significantly affect processing speed?

jantuitman, 2023-06-24:
I started working on this.
 
 The draft PR is: https://github.com/SynthstromAudible/DelugeFirmware/pull/102

sichtbeton, 2023-08-07:
lets close the issue then.