SynthstromAudible/DelugeFirmware #55: Piano keyboard

stosem, 2023-06-14:
In addition to the isomorphic keyboard, with a 16x8 button matrix, we can create an 8 octave piano keyboard.
And bind this piano keyboard to the second click of the "Keyboard" button (or switch keyboard type in the Settings menu).

litui, 2023-06-14:
Is this a feature suggestion or a commitment to work on this item?  Feature suggestions generally go in the Synthstrom Forum.

bfredl, 2023-06-15:
Would this be addressed by using #46 and setting the row spacing to 12 semitones?

stosem, 2023-06-15:
It may looks like this. So, dynamic population is possible, using loops and intervals, but seems to have an additional logic. [image]

PaulFreund, 2023-07-01:
@stosem Do you still want to work on this? I'm planning to start work on a PR with a multi layout (piano, isomorphic, fixed), configurable (rows, columns, in-key only, chords) keyboard that should implement #55 and #19 extending on #46 and don't want to collide with anybody

stosem, 2023-07-01:
> I'm planning to start work on a PR with a multi layout 

That would be nice! Yes, start please, I'll check you MR and maybe join your work with testing or any other help. Thanks!

PaulFreund, 2023-07-28:
Hi @stosem, the Keyboard Layout API is done and merged into community. Maybe give it a look and tell me if you would like to implement it or if I should do it :)

sichtbeton, 2023-08-07:
this could be close IMHO. either make a PR out of it or close it. its not an issue but rather a feature suggestion.