SynthstromAudible/DelugeFirmware #70: Next steps for DBT

litui, 2023-06-19:
Now that DBT is in `community` there is a laundry list of stuff I want to add to it but I'll start with the following:

- [ ] Build stats on completion, sizing, etc.
    - [ ] Danger-checking/warning for .bin size 0 or oversize .bin
- [ ] Dynamic version string (eg: "c20230619-a1b2c3d" in the scheme: &lt;c for community&gt;&lt;yyyyMMdd&gt;-&lt;git_commit&gt;-&lt;build type: "r" or "d"&gt;)
- [ ] Direct target support / multitarget (eg: `./dbt dbt-build-debug-oled dbt-build-release-oled` etc.)
- [ ] Debugger shortcuts
    - [ ] DelugeProbe and FTDI compatible openocd + gdb shortcuts
    - [ ] JLink support with whatever tools they use (know very little about JLink, but have some starting points from over at the Flipper Zero repo)
    - [ ] Quick, commandline "install" support via debugger-based hardware flashing

More to be added.

m-m-adams, 2023-06-19:
Copying from the PR - on ARM Fedora 37 I get an error about missing libcrypt.so, but it builds successfully after installing libxcrypt-compat. This is arguably a distro issue but probably worth noting it as a requirement since this appears to affect Arch and WSL2 as well per this discussion in python-build-standalone: https://github.com/indygreg/python-build-standalone/issues/173

Otherwise I guess it will go away once everyone is running python 3.13

litui, 2023-06-19:
I'll add a bullet for that. I was thinking about the best way to approach testing for that and providing a notification if it's missing. Maybe something like:

 **if** the OS is Linux, grep `ldconfig -p` and determine if libcrypt.so.1 is in place/what we're expecting. Could even maybe ldd the file if we need to dig deeper.

eg `ldconfig -p | grep "libcrypt.so.1"` output
```
libcrypt.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libcrypt.so.1
```

bfredl, 2023-06-19:
> This is arguably a distro issue but probably worth noting it as a requirement since this appears to affect Arch and WSL2 as well per this discussion in python-build-standalone: https://github.com/indygreg/python-build-standalone/issues/173

A possible solution could be to use the system's `python3.11` executable if it is available, and set up a venv to still install the packages into an isolated environment.

litui, 2023-06-19:
Possible, not my preference though. Venvs are useful, but that's a whole other pile of logic and additional layers of path mangling magic I'd want to avoid for simplicity's sake. In one of the earlier iterations of DBT I tried bundling poetry wheels and using the system python, and it was messier than I'd like.

litui, 2023-06-20:
Not done with this issue yet!