SynthstromAudible/DelugeFirmware #84: Snap-to-zero-crossing in

bobtwinkles, 2023-06-22:
From a discussion on Discord earlier today: https://discord.com/channels/608916579421257728/1118654563596128336/1121185820854980678

In the waveform editor, the following shortcuts should work:
- Vertical scroll encoder should swap the currently selected endpoint in the order
  1. start
  2. loop start if pulled out
  3. loop end if pulled out
  4. end
- Holding down the select encoder while moving it should snap to zero crossings

weavermedia, 2023-06-22:
Could we also add a highlight to the zero crossing row in the waveform display? Probably only really useful when zoomed all the way in to single-sample level. I do single cycle looping on samples sometimes and having the zero crossing sample column brighter (or another color) than the others would be helpful.

sichtbeton, 2023-06-30:
and maybe to have it backwards compatible, this could be made optional as a "community setting".

weavermedia, 2023-07-21:
Looked into zero-crossing a little. There's a common method for finding where the audio sample level crosses the 0 line which just compares the sign of the integer. When the sign changes, that's a zero-cross point.

But we have to remember we're dealing, for the most part, with stereo samples. We probably can't just look for zero-crossing because for both stereo channels to cross zero on the same sample is far less likely.

Maybe a combination of doing an average of both channels with some low threshold might work?

Probably best to actually find out how other systems do it. Maybe look at open source DAWs and/or sample editors.

sichtbeton, 2023-08-07:
maybe this could be turned into a discussion as its not really an issue rather then an enhancement?

weavermedia, 2023-08-08:
@sichtbeton good call 👍

closed 2024-12-28 by m-m-adams