SynthstromAudible/DelugeFirmware #90: Missing USB MIDI devices

drbourbon, 2023-06-22:
Just build from origin/community and noticed that external MIDI devices connected via USB are not visible under Settings -> MIDI -> Devices.

They are visible with last official 7-segments Synthstrom firmware V4p0p1 and with firmware released by @alter-alter in his fork here https://github.com/alter-alter/DelugeFirmware/releases/download/1.0.5_randomizer/Deluge-release-7seg1051.bin

Steps to reproduce (darwin, macOS 12.6.5):
`git clone https://github.com/SynthstromAudible/DelugeFirmware && cd DelugeFirmware && ./dbt dbt-build-debug-7seg`

m-m-adams, 2023-06-22:
I'll troubleshoot later today - I suspect the hosted usb midi device structure isn't getting properly initialized with the change to have an array of virtual ports per device

m-m-adams, 2023-06-22:
@drbourbon to help me narrow it down could you confirm if the problem is just that the hosted devices can't be seen in the menu, or are they also not sending/receiving midi?

drbourbon, 2023-06-23:
> @drbourbon to help me narrow it down could you confirm if the problem is just that the hosted devices can't be seen in the menu, or are they also not sending/receiving midi?

I just tried and I can't send nor receive midi (tried with a standard MIDI device and with a MPE device)

m-m-adams, 2023-06-23:
> > @drbourbon to help me narrow it down could you confirm if the problem is just that the hosted devices can't be seen in the menu, or are they also not sending/receiving midi?
> 
> 
> 
> I just tried and I can't send nor receive midi (tried with a standard MIDI device and with a MPE device)

Could you try building from the PR to fix this? I believe I've got it working now, no issues hosting my linnstrument

drbourbon, 2023-06-24:
It's working now, thanks

closed 2023-06-24 by drbourbon