SynthstromAudible/DelugeFirmware #92: Individual track export

trappar, 2023-06-23:
This seems to be the most requested feature from users. The ability to have Deluge export individual tracks would make it far less tedious to take a song and finish it on a computer. I've seen users suggesting different methods for this, so I'll describe a couple and maybe this can serve as a basis for discussion around other ideas and which would be best to implement.

## 1 - Batch track export

This would be an automated way of recording each individual track in a song. I'd imagine the process as:

1. Select a track and time range for export.
2. Confirm selection.
3. Deluge automatically plays though each track, recording in realtime.

### Notes
* Ideally with this option the files should be in some way colocated with the song and named appropriately like `[song]-[track].wav`.
* Running this could take quite a long time to complete, but it would still be far better than having to do all this work manually.

## 2 - Track freezing

Another option would be to allow for freezing of tracks. It's possible that this would be the same as the prior option at its core, but once a track was frozen, playback would use the frozen version instead of the live one - allowing a user to save on CPU / Ram.

Additionally it would probably be helpful to allow for batch track freezing. I could imagine a UI like:
1. Each track becomes individually selectable using pads.
2. Confirm selection.
3. Deluge records each track.

### Notes
* The frozen versions would need to be saved along with the song similarly to how it would work with batch track export.
* There would probably need to be some way to visually identify frozen tracks.
* What would the process look like for unfreezing a track?

## 3 - Deluge as a multi-track audio interface

This option is the most nebulous to me. If it were possible to have Deluge act as a multi-track audio interface, then maybe users could simply record directly into their DAW. I'm sure an option like this would have limits. Some questions:

* How many tracks could Deluge simultaneously offer?
* Would there need to be a way to configure which output each track would be sent to?

## Additional thoughts

I'm not sure if the offline nature of this options 1/2 grant any possibility for non-realtime rendering, but it's probably something that's worth looking into. Maybe there are optimizations that could be made like detecting silence in tracks and then creating a silent region of the same length as the remaining time until the next audio starts and skipping to that point. I'd imagine that batch export would be painfully slow without some optimization like this.

Would love to hear back from active devs on what seems like it could be possible here!

bobtwinkles, 2023-06-23:
Thanks for writing this up! I'm interested in implementing track freezing in the next couple of weeks, time permitting. What would really help is having a collection of song files, of varying complexity, to use as a basis for testing. I'll post a message in the discord asking about that as well.

m-m-adams, 2023-07-30:
Offline rendering is tricky to implement, the deluge audio engine makes lots of assumptions on time based on how frequently it's called. A way to automatically bounce tracks one at a time might be the best way to go for now, it'll at least save the manual process of resampling