SynthstromAudible/DelugeFirmware #107: Shift Lock

bobtwinkles, 2023-06-26:
A few possible options for implementation:

 - Menu setting to turn shift in to a toggle (probably need this regardless
 - Tap shift 7 times for sticky keys (it must play [the sound](https://www.myinstants.com/en/instant/sticky-keys-blip-59783/))
 - Midi CC learn (this is a good stretch goal, but may be more technically complex)

weavermedia, 2023-06-28:
Is this to keep shift enabled after letting go? What about double tap or long press?

bobtwinkles, 2023-06-28:
> Is this to keep shift enabled after letting go?

That's correct.

> What about double tap or long press?

I'll make the number of taps required to enable/disable shift lock configurable. Double tap seems too easy to do by accident during performance for it to be a default setting. Long press similarly will likely collide with people's muscles memory (I know I hold shift for long enough to be considered a "long press" regularly). I'm not opposed to it being implemented, but I'm unlikely to do it myself.

weavermedia, 2023-06-28:
I also hold shift for long times regularly when programming synth sounds. That's why I think long press would be good. Just hold for a couple of operations then let go and carry on with shift operations. Then just a quick tap on shift to turn it off.

I'm only talking about studio work. I don't of any live stuff so can't speak to that, but I'm pretty sure I wouldn't want to be doing shift operations on the Deluge live anyway - too prone to error. I'd prefer to map that to a MIDI controller.

ok-reza, 2023-06-28:
From a performance perspective, you definitely don't want to accidentally Shift lock. For instance, it would disable any use of global 'Affect All' effects in SONG view. I generally am constantly holding SHIFT while performing in SONG view since I launch clips immediately often, so long press wouldn't work in my case.

This is all to say a larger number of repeated taps would be safer, 5-7 depending seems like a good number.

I wonder if it makes sense to have no other button pressed when executing the 5-7 taps in order to avoid accidental Shift Locks?

bobtwinkles, 2023-06-28:
> I wonder if it makes sense to have no other button pressed when executing the 5-7 taps in order to avoid accidental Shift Locks?
This is probably doable. If you want you can try one of the builds in #118 and see if it feels necessary. For now I've implemented only multi-tap, but I'll work on long press tonight if I have time.

bobtwinkles, 2023-06-29:
Everything except the sound and MIDI CC learn is implemented now, please use the builds from #118 to test.

sichtbeton, 2023-08-07:
could this be closed as there is a draft PR #118 ?

PR #118 (2023-09-02): Sticky keys
Fixes #107 

TODO:
  - [x] Long press enable mode
  - [x] Illuminate shift when it's stuck on by sticky keys
  - [x] Audit the various test modes to make sure they will work with stickykeys (either by disabling it or adding appropriate calls to the shift mode code)
  - [ ] Custom UI class for the sticky keys menu (7seg compat + long press enable/time configuration)
  - [ ] Add audible indication when sticky keys enabled (if it's easy, this is a very silly feature)
commit 9e4254af (2023-09-02): feature: sticky keys (#118)

This implements a "sticky keys" function for the shift key, which
toggles on short press. It also wires up the shift LED to give feedback
when sticky keys is enabled. This can be overriden in case the user
doesn't want it (and can be enabled separately if they want feedback
even when sticky keys isn't enabled).

Fixes #107
commit 010c4f82 (2023-09-13): feature: sticky keys (#118)

This implements a "sticky keys" function for the shift key, which
toggles on short press. It also wires up the shift LED to give feedback
when sticky keys is enabled. This can be overriden in case the user
doesn't want it (and can be enabled separately if they want feedback
even when sticky keys isn't enabled).

Fixes #107
closed 2023-09-02 by PaulFreund