SynthstromAudible/DelugeFirmware #109: Inactive pads randomly flash when sequence is playing

ok-reza, 2023-06-26:
I've observed this behavior using the most recent community build for 7SEG. (June 26, 2023)

Video demonstrating bug: https://www.youtube.com/watch?v=lYFI56dkVE8
(Skip to 1:00 for another way to reproduce the bug via TEMPO knob)

### How To Replicate
1) Press Play
2) Go to Keyboard View
3) Repeatedly press one note
4) Observe multiple inactive pads light up

Alternatively:

1) Press Play
2) Enter SONG view
3) Turn TEMPO knob up.
4) Observe multiple inactive pads light up.

Sometimes pads will stay LIT until you leave the view.

This does not occur with the initial Community Build 7SEG firmware.

bobtwinkles, 2023-06-27:
We (@stellar-aria, @m-m-adams, and myself) did some debugging of this on Discord and narrowed this down to this change: https://github.com/SynthstromAudible/DelugeFirmware/commit/fd4b6f606937166a8b3affd66d630d5cc761fe53

It appears any minor change to those routines (even something as simple as moving the *exact same code* to an `__attribute__((always_inline))` function) seems to destabilize the PIC communication channel. I'm going to revert the commit above for now, but we should keep this bug open as a place to track ongoing investigation in to what is going on with the interaction between the PIC/UART and the OLED routines.

PaulFreund, 2023-06-28:
I did some digging and right now think it might be some "fast" mechanic for rendering the transport position (play position) or similar writing in between normal keyboard Rendering. I didn't find it yet. Tested with fd4b6f6. I might want to test with it's parent or look at the disassembly diff because with O2 at least in godbolt it emitted the same code.

PaulFreund, 2023-06-30:
Problem was very easy to overlook, see https://github.com/SynthstromAudible/DelugeFirmware/commit/fd4b6f606937166a8b3affd66d630d5cc761fe53#diff-12f405c510dad3614b5b13d025daef00c3711cbfda22185a5c9186c07aa6cd9cR91

PaulFreund, 2023-07-02:
@alex-r-t can you still reproduce with the current community build?

closed 2023-07-30 by m-m-adams