SynthstromAudible/DelugeFirmware #113: Deluge won't boot from SD since 884da39

litui, 2023-06-28:
Using 7seg deluge, tested before and after this merge. Loading firmware from SD card resulted in unresponsive Deluge on commits after 884da398a3b622bdab414b420041cac867b26008.

@stellar-aria has a hotfix in the works.

litui, 2023-06-28:
Reopening issue as work on the bug gets resolved, but @bobtwinkles 's PR reverted the change so this is no longer urgent.

stellar-aria, 2023-06-28:
The problem was discovered to be two bugs: one, in the linkerscript with ordering and offset, and the second (much more complicated one) being related to init order of global variables, which is technically undefined behavior.

As of current work at 8d402b713a495cc20a4f2ebd1a45c1909b6a58ed, the linkerscript is confirmed working and acting as normal.

Next steps involve slowly removing manual initialization in favor of global initializers until we find the culprit out-of-order initializer that is causing E140.

PR #117 (2023-06-28): Linkerscript rework and global initializations - Attempt 2
Despite the builds working completely fine over hardware debuggers, @litui found an issue where the alignment and positioning of certain linkerscript sections made it so the deluge would not boot when the firmware was written to the QSPI ROM.

This is and should only ever be an issue with the linkerscript, so at least now we know the root cause, (and honestly should never have to edit the thing again).

There seemed to be an issue with initialization order (from my panicked attempts at fixing things last night), but methodically updati [... cut]
closed 2023-06-28 by jamiefaye