SynthstromAudible/DelugeFirmware #126: Kit view items can't be triggered simultaneously

sichtbeton, 2023-06-30:
Steps to reproduce:

- kit with a sample row
- enter kit 16 levels kit view
- trigger one of the velocity buttons and hold it down
- trigger another velocity button from the same kit item

Expected behavior would be to hear a second sound. But it just stays silent.

It would be preferable to hear the second trigger too, making finger drumming much easier and accessible.

m-m-adams, 2023-07-01:
I looked at this when I was integrating the mode but one sounding note per row is pretty fundamental to the deluge sound engine - consider what the sequencer would look like if you had two overlapping notes. It would be possible to have a second note cut off the first which might be more intuitive for percussive sounds

weavermedia, 2023-07-01:
I always wondered why kit sounds had the POLY and LEGA options in the VOICE menu when they don't actually do anything. Or do they?

For the kit view my preference would be for the second note to cut off the first note rather than not hear the second note.

sichtbeton, 2023-07-01:
> I always wondered why kit sounds had the POLY and LEGA options in the VOICE menu when they don't actually do anything. Or do they?
> 
> For the kit view my preference would be for the second note to cut off the first note rather than not hear the second note.

Actually, this is a very cool idea. This way you can play very rhythmical with longer samples.

weavermedia, 2023-07-04:
Any reason why this is not planned?

sichtbeton, 2023-07-04:
I guess nobody was interested in this. Can leave it open of course.

bobtwinkles, 2023-07-04:
In general leaving the issue open is preferred, as it helps track what
people are interested in and a venue for collecting technical discussion
related to implementation or bug fixing.

m-m-adams, 2023-07-05:
Agreed - while this is currently 'intended behaviour' I do think the discussion around how duplicate notes are handled is worthwhile and I'd love to see a creative solution for it

The same issue also applies in normal synths, if you hold a note and then press the same note elsewhere on the keyboard nothing happens

sichtbeton, 2023-07-18:
So should I close this issue then as it seems to be intended?

m-m-adams, 2023-07-30:
Moving to discussion for now - I think an option to cut off existing notes when a new note on is received is reasonable if someone wants to implement it