SynthstromAudible/DelugeFirmware #130: 4-op FM Synth expansion

m-m-adams, 2023-07-01:
The FM synth could be expanded to allow selecting between waveforms rather than just sine. In order to do this without aliasing the maximum frequency generated by the underlying oscillator must be able to be configured dynamically, using a method such as blep/blit. 

The maximum frequency cutoffs for an oscillator to avoid aliasing can be calculated by inverting the methods in chownings 1973 paper. 

Some filters would have to be enabled for this to be useful - the SVF or a simple LPF without drive would minimize the impact on voice count

alex-r-t, 2023-07-01:
Here's some thoughts/ideas:
- Would using wavetables as carriers/modulators be feasible, or would that incur too much of a performance overhead?
- How much of a rework/an optimization issue would allowing for different orders/algorithms be?
- Would through-zero FM be a possibility?

m-m-adams, 2023-07-01:
1. Might be - wavetables are already reconstructed using an ifft for band limiting, I'd have to look at the performance impact of setting up wavetables more frequently

2. Algorithms would be tricky with the UI (especially 7seg) but allowing each modulator to be sent to carrier 1/2 and the other modulator might be doable 

3. It's already linear+through zero, allowing moog style exponential non through zero cross mod in subtractive mode might be cool though

bfredl, 2023-07-01:
More envelopes would be great with FM, either yamaha-style with a wired envelope per operator, alternatively rework of the mod matrix to allow more envelopes overall (which would be worth it for subtractive as well).

litui, 2023-07-04:
I think different algorithms would be cool, but perhaps limited to a finite list of known-to-be-reasonably-efficient algorithms unless somebody wants to go ham on an "algorithm editor" screen or set of screens.

bfredl, 2023-07-04:
An approach for more algorithms could be to extend the destination menu for mod2 (i e choose OSC1 and OSC2 separate and not just both) and implement it for mod1. 

Unfortunately "destination" takes the same row with "wavetable" which would be a problem for using wavetables for the mods (or to add add a "destination" menu for OSC2 to change it to be yet a modulator, which would be needed for an op chain like mod2>mod1>osc2>osc1 ). So we might need to add a community option which changes around the shortcut layout for the FM synth.

m-m-adams, 2023-07-22:
Going to reopen this as a discussion since I'm not going to work on it until I finish with filters

closed 2023-07-22 by m-m-adams