SynthstromAudible/DelugeFirmware #133: Notes held before recording starts aren't captured in recording

trappar, 2023-07-01:
When using the keyboard or drum keyboard, a pad that's being held before recording starts will not be included in the recording. This is especially relevant when playing in mono/legato patterns where you might hold a low note while alternating over higher notes.

To replicate with the synth keyboard:
1. Arm recording
2. Hold down a note
3. Press play
4. Let go of the note before the clip finishes
5. Observe that no note exists in the recorded clip

It's basically the same with the drum keyboard, but in that case it probably only makes sense to try to catch sounds which are using the cut, loop, or stretch modes.

weavermedia, 2023-07-02:
If anything was implemented to capture existing held notes when recording starts I hope it would be optional because the issue you described sounds like intended behavior to me.

I can see the desire to capture existing held notes but it's unusual enough behavior that I think it might surprise most people.

litui, 2023-07-04:
I would agree with @weavermedia. While this behaviour is intuitive as audio-recording behaviour, it's not something I've seen in the midi world previously. I can't even imagine how it could be done particularly well.

weavermedia, 2023-07-05:
Can this be closed as 'intended behavior'?

sichtbeton, 2023-07-05:
agreed, its as expected in MIDI