SynthstromAudible/DelugeFirmware #134: Old Bootloaders (<002) can not boot

bobtwinkles, 2023-07-02:
After #117, most Deluges can boot. However, the oldest bootloader revisions do not seem to be able to boot. @alex-r-t has kindly provided some builds ([on discord](https://discord.com/channels/608916579421257728/1123814461745156167/1124848003413442612)) which differ only by the linker script. Using Rohan's original linker script, which places `.bss` before `.reset`, works on very old bootloaders, while current top of tree (which places `.reset` before `.bss`) does not.

bobtwinkles, 2023-07-02:
alex also provided a [dump of the v0 or v1 bootloader](https://discord.com/channels/608916579421257728/1123814461745156167/1124189338415157298). Reverse engineering this shows that the bootloader splats the `.bin` file directly to flash, at offset `0x80_000` on both the v0 and v7 (provided by synthstrom) bootloaders. The bootloader loads the following offsets in to that image:

- Offset `0x20` is the RAM address to start copying to.
- Offset `0x24` is the RAM address to stop copying at.
- Offset `0x28` is the address to jump to after boot.

The V0 and V7 bootloaders differ in their termination conditions for the copy loop, this likely merits further investigation.

jantuitman, 2023-07-02:
ah, that might explain the issues i walked into this morning. i had merged the community branch into my development branch and now i cannot boot, so i had to revert back to my development branch without merges of the community branch. hope someone fixes this soon, because i cannot incorporate latest changes in my code now.

bobtwinkles, 2023-07-02:
Do you have a working debug setup, with some way to signal how far you've
gotten in the boot process? I'm working on debugging this but it's purely
by code inspection, so someone (or multiple people!) attacking this from a
dynamic perspective would be immensely helpful.

We think the problem is related to instability in the initialization of
global objects, so demonstrating we're getting to main/deluge_main (or if
not, what's causing the fault) would be nice.

jantuitman, 2023-07-02:
I am sorry, I have no experience with embedded software before, linker scripts, and boot loaders. I also dont have anything attached to any UART port of the Deluge, I just make release builds with command line SCONS on the mac. If i see what is happening in the #117 PR I dont see anything i could contribute.

weavermedia, 2023-07-04:
I don't have a debug setup but just in case it's useful this is my experience:

1. Build from source, latest commit [db282c06b1bd6a5caba14b9e1e1cbd0c00725ccf](https://github.com/SynthstromAudible/DelugeFirmware/commit/db282c06b1bd6a5caba14b9e1e1cbd0c00725ccf)
2. Put 7-seg bin file on SD card, insert card into Deluge
3. Hold SHIFT + power on
4. Display shows UPDA for a second
5. Display shows slow spinner for 3 seconds then fast spinner for 23 seconds
6. Display shows DONE but Deluge is unresponsive
7. Power off then on and Deluge is unresponsive

I realize that last step is where you need debug detail and I wish I could help but this is all I got.

weavermedia, 2023-07-05:
Status update - currently working with @stellar-aria testing a bunch of linker scripts on a 0001/0002 bootloader Deluge to see what's breaking.

I'm doing the easy bit of compiling and updating firmware. Not the hard bit of finding out what the problem actually is 😜

closed 2023-07-06 by jamiefaye

stellar-aria, 2024-09-10:
> Hi and thank you for doing this! Hope it's okay to ask this here as opposed to Facebook...I have a Deluge numeric currently on bootloader 0005 and would like to update. Does this mean it is now (really really) safe to do this within community firmware?

There's now no need to update the bootloader to support community, and we've actually removed the functionality to do so from the community firmware to prevent it from potentially bricking Deluge devices.

sinewave440hz, 2024-09-10:
Thank you very much for clarifying that :)

PR #159 (2023-07-06): Fix .bss position and put .text inside .reset
After much testing and checking, we have finally found the two changes that make the firmware work on older bootloaders again.

This closes #134.