SynthstromAudible/DelugeFirmware #142: Voice dropouts FW 3.15 vs 4

sichtbeton, 2023-07-03:
Stress test of the Deluge comparing FW 3.15 and FW 4 (community)

Testing setup

- Deluge 7seg
- original SD card 32gb formatted with mac OSX 32kb cluster size MS DOS FAT
- newer deluge model (no old bootloader)
- 2 firmwares (original 3.15 from SA / FW 4 community edition from 03072023)

Test

- flashed the Deluge to FW 3.15
- made a song with long releases and too much tracks (lots of arps + delay + unison)
- resampled reference song in 3.15 playing through with no dropouts
- refreshed the Deluge with newer FW 4
- opened the same song and resampled playback

Result: 

- newer FW 4 keeps dropping voices with the same song
- especially audible in the longer releases of the keys (1st clip top)
- dropouts variable in length
- 3.15 plays stable

all files can be found here:
https://drive.google.com/file/d/1-AR-5WxnJaHFXip0Gsh2bOirj1FZE5I0/view?usp=sharing

sichtbeton, 2023-07-05:
> Is this just a symptom of the extra processing required by the new features merged into the repo? Or do you think it's a bug?
> 

This has been going on since fw 4. so it must be something deep inside the code when changing from 3.15 to 4. 

> 
> Do you have any metrics for how many/how early voices are dropping?
> 
> 

It's mostly audible in the long releases of the pads in clip 1 on top. Maybe more, but it's hard to say.

No metrics because there is no easy way to measure this. Also no cpu meter to gauge whether this is a cpu thing after all.

> 
> Do we know which, if any, new features are likely to cause heavier CPU load and/or increased SD reads/writes?

It was introduced in 4. Rohan just released a new fw 4.1.4 beta 1. but it's the same.

Question: can you or anybody else replicate this or am I simply going crazy?

weavermedia, 2023-07-06:
We could do with a more structured way to find when voices are cutting out.

A simple approach could be play a sustained C2 then just start stacking more sustained notes on top on 16ths, starting at maybe C4. When you hear he C2 cut out you have a point in time and a max voice count to work with. Then if that's reproducible it might be considered a solid metric.

sichtbeton, 2023-07-06:
Thing is: different things require different CPU resources. 

Testing only one type of sound and modulation could easily overlook any other relevant roadblock in CPU usage. I do agree that a more methodical approach would be good. 

Can’t think of one right now that would include most of the relevant Deluge features.

weavermedia, 2023-07-07:
Yes, agreed it would be difficult to include every Deluge feature in a s ingle test. What I'm proposing is the other way around. More of a simple base stress test of voice count. Then to test a particular feature you would add that to the test.

I made the test song that I described and it's quite accurately reproducible on every playback. The voices start stealing at the point every time. If I add reverb they steal earlier. If I add delay it's earlier still, change delay to analog and earlier still, etc.

Plus a simple voice stealing test like this seems like it would be good for testing between versions.

sichtbeton, 2023-07-08:
Can you upload this song?

m-m-adams, 2023-07-09:
Couple things to add:

1. Voice dropping is recorded over RTT. Approximately 40-50 init patch voices can play at once, the number of voices and impact of different DSP changes with optimization levels

2. No current features have a significant impact. The SVF is much lighter on DSP than the ladders (70ish init voices), and the master compressor just reduces the count by 1 or 2. 

3. A change could just be differences in compiler settings as dbt uses a newer GCC than Rohan used. We may want to try O3 optimization for release but it will require testing. Int overflow (UB) can change at O3 and I believe that can occur in this codebase

sichtbeton, 2023-07-15:
> Couple things to add:
>
> 1. Voice dropping is recorded over RTT. Approximately 40-50 init patch voices can play at once, the number of voices and impact of different DSP changes with optimization [...]

This has been going on since fw 4 supplied by SA. So it has nothing to do with the community fw.

m-m-adams, 2023-08-04:
I'm going to close this for now - without having any idea where in the code the performance impacting change occurred I don't think we can meaningfully investigate this 

Of course looking for spots to optimize the code is always useful but that's not really a bug

closed 2023-08-04 by m-m-adams

sichtbeton, 2023-08-05:
I disagree. Having a song play fine with 3.15 and not with the recent fw is a clear bug to me. But ok.