SynthstromAudible/DelugeFirmware #179: Svf filter cuts too much high frequencies in default position

sichtbeton, 2023-07-10:
The new svf filter is great. The only issue I found is that when engaged, the filter cuts a lot of high frequencies by default. So doing a filter move is not really possible because there is no way to get the original sound when svf is selected.

m-m-adams, 2023-07-19:
@sichtbeton can you test a new build and confirm whether this is now good enough? It's still off by a factor of sin(f)/f compared to the ladders so it matches almost exactly at low frequencies but is at about 17khz instead of 22khz fully open

sichtbeton, 2023-07-20:
> @sichtbeton can you test a new build and confirm whether this is now good enough? It's still off by a factor of sin(f)/f compared to the ladders so it matches almost exactly at low frequencies but is at about 17khz instead of 22khz fully open

Ok, checked it just now. It's much better now! There is still a audible difference when put under the microscope (DT770 Pro headphones), so it's loosing a bit of the high freqs when fully open compared to 12db / 24db filters. "Drive" reacts completely different (louder than the rest) so I would ignore this. 

So would most users notice it? Probably not and if so only the nerds. Could there be a bit more high freqs when fully open? Sure.

m-m-adams, 2023-07-20:
Makes sense - I'll try to address it more completely as I get through #105

m-m-adams, 2023-07-30:
Closing for now - will be done in #105

closed 2023-07-30 by m-m-adams