SynthstromAudible/DelugeFirmware #195: Pan Parameter Value Display on OLED

seangoodvibes, 2023-07-14:
Discovered an issue with how the **Pan** parameter is being displayed on the OLED screen, both when changing the param using the select knob and the gold knob. See attached picture.

I am using the latest community build.

To replicate bug:

1) enter a clip
2) press shift + master pan
3) turn gold pan master fx knob or turn the select knob to change the parameter value

[image]

m-m-adams, 2023-07-20:
Confirmed this is still happening

m-m-adams, 2023-07-31:
The problem seems to be that pan.h is getting the limits from integer.h in menu_item instead of the integer.h in patched_param. This sets the limits to be 0-50, and there is int overflow when the value goes above 32

closed 2023-07-31 by jamiefaye