SynthstromAudible/DelugeFirmware #203: Gold knob unresponsive to quick turns

balcombits, 2023-07-16:
When quickly turning a gold parameter knob (ex: changing HPF frequency (see video attached)) the parameters hang and/or jump forwards/backwards randomly by incremental amounts. 

Running Deluge-release-7seg108-with-community.bin firmware by user: alter-alter

Video: https://youtu.be/FNYsKK-ZE5w

reproduced by users on the official discord.

m-m-adams, 2023-07-16:
Reproducible on both official and community releases while the sequencer is running and synths are playing. I suspect that encoder increments are dropped if multiple are received before the next call to Encoders::readEncoders(). There is an interrupt function in encoder.cpp but it does not appear to be used

balcombits, 2023-07-16:
It’s been determined this is hardware related to my specific device.

closed 2023-07-16 by balcombits