SynthstromAudible/DelugeFirmware #217: no scrolling on 7seg display

sichtbeton, 2023-07-20:
for now, the scrolling of the 7seg display is not working anymore. sure, it was distracting to repeat the scrolling forever.  but having not even one scroll initially and being forced to scoll manually makes the display even harder to read then with scrolling.

I suspect this behaviour was introduced in either of these PRs:

https://github.com/SynthstromAudible/DelugeFirmware/pull/176
https://github.com/SynthstromAudible/DelugeFirmware/pull/178

> I also disabled OLED text scrolling because i find it incredibly distracting. I simply commented out a line in View.cpp.  

Suggestion: 

- revert to the original scrolling by default
- have an option to scroll only once and then stop
- another option to never scroll
- additionally: scroll speed setting for slow / medium / fast

m-m-adams, 2023-07-26:
Confirmed - this should definitely be fixed, it makes 7seg extraordinarily difficult to navigate

m-m-adams, 2023-07-26:
Neither of those PRs is responsible, the change to view.cpp wasn't merged

sichtbeton, 2023-07-26:
> Neither of those PRs is responsible, the change to view.cpp wasn't merged 

This is weird. Then how come?

m-m-adams, 2023-07-26:
Who knows! It's a bug

stellar-aria, 2023-07-26:
This was done by Rohan, see:

m-m-adams, 2023-07-29:
So trying both, I think reverting this change by Rohan is an improvement for my own workflow but it does make #99 a significant issue for people who rely on the file postfix behaviour. Basically the behaviour should be:

- If we arrived at the file by typing:
    1. Show the typed in text and whatever afterwards without scrolling
- Else
    1. If the text does not end with _number, scroll normally
    2. If the test does end with _number, show the last 4 characters first but then scroll normally

I'm gonna work on a fix later today

closed 2023-07-30 by m-m-adams