SynthstromAudible/DelugeFirmware #224: Permanent midi controller global option.

Maly-Lemire, 2023-07-21:
Add an option that auto learn a midi controller (Without having to press the learn button) and automatically send to selected channel in clip mode.
I have almost never used my deluge because of that missing feature. Having to relearn every time I switch track for controlling many external synths is a pain and make the deluge unusable for live performances.

weavermedia, 2023-07-21:
Sounds a lot like this work in progress [Default midi mapping for currently selected clip](https://github.com/SynthstromAudible/DelugeFirmware/issues/216)

Maly-Lemire, 2023-07-21:
Yes sounds a lot like it! Yes or could set only to one channel in options. All I want is that when I select a clip it output on that midi channel instantly and in song or arrange mode does nothing.

Maly-Lemire, 2023-07-21:
It could also use the last selected clip in song and arrange mode if it's not to hard. I would be happy with only on the focused clip.

m-m-adams, 2023-07-22:
Hi @Maly-Lemire, welcome to the community and thanks for the feature suggestion. Could you close this issue and reopen as a discussion? We are trying to use the issues for bugs and work in progress only

Maly-Lemire, 2023-07-24:
Closed it but don't know how to re-open as a suggestion...

closed 2023-07-24 by Maly-Lemire

m-m-adams, 2023-07-24:
You should be able to open a discussion and one of the options will be feature request. If you refer to #224 in your discussion it will link to this issue

Queroland, 2023-07-25:
Hey..
sounds like you want midi output for the selected clip?
With #216 I wanted to focus on midi input first to have midi control for the selected clip - on one selectable channel. And work out a fixed midi table for that - without learning functions to keep it simple for now. And it appears not simple to me...