SynthstromAudible/DelugeFirmware #274: Align master compressor settings with Deluge UX

PaulFreund, 2023-07-31:
Currently master compressor has two pahts for the settings which are both shortcuts.

Disclaimer: For grid style session view #251 I would like to claim Shift+Pad for instant change of playback state.

On the other hand the current display menu for the compressor parameters is a departure from current look and feel of the OLED screen (while being gorgeous of course). It also masks Affect Entire Sidechain although I don't think anybody needs it. 

My current suggestion would be:
* In Session and Arranger mode while Affect Entire is off non of the FX controls are used if no clip is held
* With Affect Entire disabled use the three rightmost FX buttons together with the gold knobs similar to all the other FX parameters
* Remove the fancy menu but show the currently edited parameter in the display while turning gold knob similar to synth parameters
* Remove Shift+Pad shortcuts. They could stay in Row display (won't work in Grid display) but the manual will get more complicated (no strong opinion here though)

m-m-adams, 2023-08-02:
Another possibility is just improving the deluge sidechain instead of maintaining two compressors. A possible approach is:

1. Feed sidechain compressor with each clips RMS level post-fx, calculated when it's copied to the song buffer 

2. Enable sidechain from synth clips and audio clips 

3. Change song level affect entire sidechain knob to increase send level and sidechain->level amount (e.g make it a one knob compressor)

weavermedia, 2023-08-03:
@m-m-adams I like having a 'proper' compressor with independent controls for threshold, attack, release, ratio and makeup gain. Feels like a step backwards to take those away now.

It's a nice bonus to have the mix control too, for parallel compression.

m-m-adams, 2023-08-04:
It could stay in the community feature menu or remain editable in more detail via the menu system. I really feel like it's a poor fit for the deluge UI by default, especially since it's exclusively a master effect - you can't send it different levels per track. 

Technically the only thing you would lose in my suggestion is makeup gain - the sidechain envelope already has controls for threshold (shape), attack, release, and ratio (vol duck). I could add auto makeup gain and then the level can be fine tuned with the affect all level

weavermedia, 2023-08-04:
I'd be very happy for the master compressor editing to take place in the menu system, especially if this also brought about editing access to all Affect Entire settings in the menu system.

Definitely interested in hearing an improved sidechain as you described but I'm not convinced it would sound as good as the master compressor.

Personally I like them being separate - a dead-simple sidechain for basic ducking and a full features master compressor for the main output. Long term I'd also like to see different compression responses like VCA, opto, transparent, etc.

m-m-adams, 2023-08-04:
To be clear all compressors use a sidechain to trigger their envelopes, there's no reason exposing the compressor envelope as a modulation source should lead to worse compression

I'll be changing the side chain envelope from triggered to an envelope follower anyway, it would be possible to just use the master compressor envelope instead if you think it's better

weavermedia, 2023-08-04:
Yeah, that's what I'd like to hear - the current sidechain feature triggered by audio levels rather than note events.

Not to quibble over terminology but not all compressors use a sidechain to trigger their envelopes. Many compressors don't even have a sidechain at all. They simply sum the input as an envelope trigger.

The term sidechain is commonly used when exposing that summed audio signal to the outside world. This allows things like HP filtering to prevent the bass from killing the mix, or replacing the audio trigger with an entirely different trigger source like a kick drum.

The Deluge sidechain compressor feature is more like those VST/AU plugins that simply duck the audio input in response to an internally generated trigger synced to host tempo. Plugins like KickStart, STFU, LFOTool all do this. They're probably more accurately described as volume shapers than compressors since they don't use the input audio as a trigger, which is a key element of a traditional compressor.

m-m-adams, 2023-08-04:
Compressor implementations are written the other way around, all compressors feed a sidechain signal into an envelope generator but some sum all their inputs into the sidechain

[image]

Yes the deluge's current sidechain is a triggered envelope and not an envelope follower, that is what I proposed changing in the first bullet of my comment

bobtwinkles, 2023-08-04:
I think there's a place for both a proper envelope-follower based compressor and the existing A/D envelope volume duck effect. I know people do use a "ghost" kick track that intentionally has no sound -- this is useful when e.g. you want to mute the drums but you still want the pumping effect in your bass line. I imagine the A/D envelope is also cheaper in terms of CPU than a proper envelope follower.

m-m-adams, 2023-08-04:
Yeah I'll put it on a toggle or something, I don't think there's space for two sidechain envelopes in the UI. Arguably you could still use a ghost kick track if the sidechain is written to ignore mutes. 

The DSP impact would be minimal since it just needs a separate compressor buffer to sum things in accordance with their sidechain send level

weavermedia, 2023-08-05:
@m-m-adams Many compressors don't feature a sidechain. They just feed the input into the envelope detector. You only need the sidechain section if you're implementing a filter or you want to expose the signal to the outside world for external processing.

[image]

It would be cool if the existing sidechain compressor could be triggered by audio with little DSP impact. I know that was a concern bandied around a lot when the code went open source. Could we conceivably see a volume shaper/compressor per clip? So we could shape drums differently to keys or bass, for example.

m-m-adams, 2023-08-05:
The envelope detector - reduction generator is called a sidechain

m-m-adams, 2023-08-05:
See also this line in the deluge master compressor (chunkware simple comp) 

https://github.com/SynthstromAudible/DelugeFirmware/blob/0eba32544b5a5a7f72abc7d67d239175097de1e8/src/deluge/dsp/master_compressor/master_compressor.h#L153

wasilios, 2023-08-29:
Hi There, 
Since i am following the compressor discussion on the main deluge forum for so many years now. I stil totaly agree that this is one of the most wanted features for the deluge. And it should get the prioritization it deserves. So i'm not sure if this is the right place for my comment. But in in my opinion the compressor topic should be an extended implementation. Not only as a master compressor. The ability to have a 3 band or 5 band compressor in trackmode (clip) would make the deluge a powerfull competitor to mashines like the sp404. 
It is an essential function especially for beat designing lo-fi beats. I hope you guys will extend the scope for the compressor functionality as much as it can be.

closed 2023-10-26 by m-m-adams