SynthstromAudible/DelugeFirmware #305: Converting between Instrument and Audio clips breaks Arranger instrument order

PaulFreund, 2023-08-04:
Either replaceAudioClipWithInstrumentClip or replaceInstrumentClipWithAudioClip or both manipulate the Clips and their underlying Outputs in a way that can lead to different ordering in Arranger and Clip view (and thus later in Grid view). Earliest tested FW is Deluge+firmware+V4p1p4-beta1+OLED.bin

m-m-adams, 2023-08-05:
I think this can also happen when you rearrange clips in song view, arranger keeps the original order. The fix might be to have arranger view dynamically resort itself to match the order of instruments in song view

m-m-adams, 2024-03-03:
Won't fix - this actually creates a new instrument so inevitably changes the order. As rows view doesn't have a real concept of tracks there's no clean way to make them match

closed 2024-03-03 by m-m-adams