SynthstromAudible/DelugeFirmware #309: sample folders of deleted songs visible on 7seg

sichtbeton, 2023-08-06:
After deleting a song that has an automatic sample folder from the "collect and save" command, the 7seg deluge still shows the folder. Either it would be deleted with the song (the best behaviour IMHO) or the folder is invisible.

m-m-adams, 2023-08-06:
This sounds like desired behaviour to me, deleting a song doesn't necessarily mean you want to delete its data

sichtbeton, 2023-08-06:
When the song is present, the folder is not. When the song is deleted, the folder is present (but empty). So it doesnt add anything, confuses only and was actually tackled by Rohan in one of his fixes. It just didnt translate to the open source version.

So its not by design and a bug.

m-m-adams, 2023-08-06:
Ah ok

sichtbeton, 2023-08-07:
> Ah ok

sorry sound too snarky. it's just an annoying thing to not be able to freely skip like it used to be possible (back in the days of 4.0.1 something).

m-m-adams, 2023-08-09:
Yeah no worries, it's not a feature I use so I wasn't sure what the expected behaviour is

sichtbeton, 2023-08-25:
This is a regression to the existing fw. I don’t think “future release” is the right category for this.

m-m-adams, 2023-09-01:
Luckily this is an open source project so anyone is welcome to fix it anyway. It's not ideal behaviour but it is easy to work around.

m-m-adams, 2024-03-03:
The samples remain in the file - I don't want to delete them, and the folder does exist. We could skip it but I'm not sure that's useful

closed 2024-03-03 by m-m-adams