SynthstromAudible/DelugeFirmware #310: shift-skipping song variations only work forward on 7seg

sichtbeton, 2023-08-06:
You can skip the song variations (2_1, 2_2, etc.) with holding shift and turning the select encoder. This works when going upwards in numbers but not when going backwards down on 7seg.

m-m-adams, 2023-08-06:
Can you verify if that occured prior to #254? I didn't know this existed and so didn't test for it

sichtbeton, 2023-08-06:
I always worked on every FW since 4. I dont know if it worked on the community FW. Adding to that: the augmentation of the versions changed, before it was with a letter (12a, 12b, 12c) now it's a numerical augmentation. (12_1, 12_2, etc).

m-m-adams, 2023-08-09:
Yeah I'm just not sure if this is a 7seg to OLED logic change or if it's a bug we've introduced. The initial community release is different than the 7seg firmware releases

m-m-adams, 2023-08-09:
Ok - if you could find the first version with the bug that would help me narrow it down, otherwise I'll get to this soon!

sichtbeton, 2023-08-25:
Currently traveling, but will do when back!

m-m-adams, 2023-10-14:
wontfix - it'll be too complex to retrofit this behaviour with folder support

PRs welcome if anyone else does it

closed 2023-10-14 by m-m-adams

sichtbeton, 2023-10-14:
> wontfix - it'll be too complex to retrofit this behaviour with folder support
> 

it did work with folders before.