SynthstromAudible/DelugeFirmware #320: Crash in Arranger preset browsing

PaulFreund, 2023-08-08:
During development of #251 I found that with current community you can generate crashes by:

1. Starting fresh (tested with debug)
2. Create a row with Kit in Song mode
3. Clone the row 
4. Change preset by holding the second row and turning the selection encoder one dent
5. Go to arranger mode
6. Hold the preset pad of the first track (rightmost) and scroll through presets with the selection encoder
7. Deluge will freze

I was not able to reproduce with current Beta official 4.1.3. If there is trouble reproducing we need to check my kits as well

m-m-adams, 2023-08-08:
Reproduced successfully

PaulFreund, 2023-08-08:
So the reason it hangs is the loop in [Song::renderAudio](https://github.com/SynthstromAudible/DelugeFirmware/blob/community/src/deluge/model/song/song.cpp#L2066) starts to reference itself. Patching done before that is in [Song::replaceInstrument](https://github.com/SynthstromAudible/DelugeFirmware/blob/community/src/deluge/model/song/song.cpp#L2974) updates the list prior to the freeze but I didn't have enough time to see if the logic is flawed or a memory issue. There is another block with sligthly different logic in [Song::replaceOutputLowLevel](https://github.com/SynthstromAudible/DelugeFirmware/blob/community/src/deluge/model/song/song.cpp#L4645)

PR #330 (2023-08-10): Fix for preset browsing freeze/crash
 - [x] Show Error I009 if replaceInstrument is called with an new instrument already in the song
 - [x] Prevent LoadInstrumentPresetUI::doPresetNavigation from generating existing instruments
 - [x] Check if other code paths can also produce I009

Fixes #320
closed 2023-08-10 by m-m-adams