SynthstromAudible/DelugeFirmware #326: Disabling 'Mod. Depth Decimals' on 7SEG doesn't disable the feature.

ok-reza, 2023-08-09:
This seems to work as intended on OLED, however on 7SEG decimals still remain.

Bin I'm using is: deluge7SEG-v1.0.0-dev+2023.08.09-72926a5 (7SEG release)

soymonitus, 2023-09-06:
@ok-reza could you re-test this with latest nightly?

ok-reza, 2023-09-06:
confirming that latest nightly (deluge-v1_0_0-nightly+2023_09_05-593bfc5.bin) for 7SEG still shows decimals when Mod. Depth Decimals is set to OFF. (test by patch filter to env1, decimals show up)

ok-reza, 2023-09-07:
@soymonitus i'm also noticing now that the osc1/osc2 transpose is missing it's decimal. if you change transpose to 1.00 it reads as '100'.

ok-reza, 2023-10-02:
@bfredl is this something that you can address? still present in latest nightly for 7SEG

PR #598 (2023-10-08): Bugfix/transpose decimal
Fix #326 (just the transpose part)

I don't know that it's worth fixing the decimals always displaying, there's no real downside and it will complicate the code significantly
closed 2023-10-11 by m-m-adams