SynthstromAudible/DelugeFirmware #327: 'Fine Tempo' defaults to ON instead of OFF when starting fresh.

ok-reza, 2023-08-09:
Since the enabled behavior is a departure from the Official Firmware, I think it makes sense for this to default to OFF.
The community_features.md describes:

> PR https://github.com/SynthstromAudible/DelugeFirmware/pull/178 - New option (FINE TEMPO in the "community features" menu) to invert the push+turn behavior of the tempo knob. With this option enabled the tempo changes by 1 when unpushed and 4 when pushed (vs 4 unpushed and 1 pushed in the official firmware). This option defaults to OFF.

So it seems to be unintentionally set to ON by default.

Handy feature btw! The only time I can think of it being useful to disable is when doing experimental AWOL deep dives/sound design craziness. Maybe begs to reason that it could be a per-project setting, but I don't think there's proper infrastructure for that just yet.

m-m-adams, 2023-08-09:
This was intentional, there's a bunch of discussion about it on the PR. We generally felt that this was a more sane default and kept the community setting around in case someone hates it

closed 2023-08-09 by m-m-adams

ok-reza, 2023-08-09:
Sounds good. From a sound design, messing about perspective, it's super handy to have the 'disabled' option, definitely don't want it taken away. But yeah, from a practical perspective it makes sense to have it enabled.