SynthstromAudible/DelugeFirmware #331: Preset browsing behavior changed compared current Beta

PaulFreund, 2023-08-09:
I did a quick check of current community against official 4.1.3 beta and behavior of preset browsing for clip with selection encoder in session seems to have changed. 

I did not find a way to change to a preset that is already in the Song in 4.1.3 beta but I am by going back and forth between two tracks in session mode in open source.

Already present in 02b83d5.

ok-reza, 2023-08-10:
double check that the preset isn't in a folder of only 1 item, it will make it seem like you can't change presets.

m-m-adams, 2023-08-12:
I can't replicate - scrolling the preset knob is correctly skipping the synths that are already in my preset

PaulFreund, 2023-08-12:
I'll do a video soon, it's not straight forward to replicate unfortunately

weavermedia, 2023-08-18:
@PaulFreund Do you mean 4.1.4 beta?

PaulFreund, 2023-08-19:
Yes, sorry got the numbers mixed up. I tested against Deluge+firmware+V4p1p4-beta1+OLED.bin. Here's the video after a fresh start with current community:

https://github.com/SynthstromAudible/DelugeFirmware/assets/678367/8a69a5b8-c400-4c6d-a4db-52de9b3c4b52

m-m-adams, 2023-08-21:
Relevant function is doPresetNavigation in load_instrument_preset_ui - I think the issue is it is called with  Availability::INSTRUMENT_UNUSED in the clip ui but not the song ui

bobtwinkles, 2023-12-17:
We should see if the fix for this is in 0501b7dc38a363f89112c1b5c36075f4c0368be9

m-m-adams, 2024-03-03:
Closing this - it's different but I don't see an issue with it, as official development seems to have stopped I think we shouldn't worry about it

closed 2024-03-03 by m-m-adams