SynthstromAudible/DelugeFirmware #342: MIDI devices menu bug

trappar, 2023-08-12:
After navigating to Settings -> MIDI -> Devices (with no devices connected) on my OLED deluge, I get a menu that appears empty, but with a weird little forward slash half way through the second line. If I attempt to scroll down nothing happens, but then if I attempt to scroll back up then what appears to be a left parenthesis appears before the forward slash.

I'm not sure what the normal behavior should be for this menu but I'm guessing it should say something like "No devices" instead of displaying random symbols.

m-m-adams, 2023-08-12:
MIDI devices should show din even with nothing else connected

Definitely a bug but I don't think it's specific to that menu, I'm getting it semi randomly on screens in the global settings menu. Closing and reopening the menu seems to fix it

stellar-aria, 2023-08-12:
@trappar Could you verify that #344 fixes this issue and displays connected MIDI devices properly?

trappar, 2023-08-12:
Yes it appears to fix this issue, but I'm still getting weird side effects that seem to be specifically associated with opening these menus. See this issue: https://github.com/SynthstromAudible/DelugeFirmware/issues/343

I can reproduced what's seen in the video from my latest comment there on a fresh boot on this firmware by doing nothing more than opening Settings -> MIDI -> Devices -> DIN Ports and scrolling around in there.

commit d01de370 (2023-08-15): Fix options menu indexing and item addition for MIDI devices (#344)

Fixes #342
PR #344 (2023-08-15): Fix options menu indexing and item addition for MIDI devices
MIDI Device menu item options weren't being setup properly (push_back instead of assignment operator), or accessed properly (passing negative offset to render function).

Fixes #342
closed 2023-08-15 by jamiefaye