SynthstromAudible/DelugeFirmware #350: Clip cloning leads to audible culling in simple songs

PaulFreund, 2023-08-13:
With very simple test song with one bassline of 16 steps (no detune) and a kit with 4 to the floor kick pattern and nothing else at 120 bpm I get audible culling and the following debug output:

```culled 1 voices. numSamples: 46. voices left: 1```

which can not be reproduced with official beta

sichtbeton, 2023-08-13:
uff. "voices left = 1"...

PaulFreund, 2023-08-14:
https://github.com/SynthstromAudible/DelugeFirmware/assets/678367/d3f23e4e-116b-4449-9ebf-7625cbc5fb18

Here is a video of the behavior with HEAD being dfa7101e

PaulFreund, 2023-08-15:
More investigation will be necessary. I started with [V4.1.4-alpha3](https://github.com/SynthstromAudible/DelugeFirmware/commit/2f6a1584c29036d8f7ca58a880146b1004dd8abc) and added which is using the original compiler and artificial bloat in main.c:

```C++
char dummy[x*1024] __attribute__((used)) = { 77 };
```

I packed the results (all Release builds) in [Inflated.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/12342096/Inflated.zip) for everyone to test. Every binary has 100k more inflation, there is also one with 350 which matches our current inflation. 

Some observations:
* After about the second to third page of copies the test above also leads to audible culling in stock V4.1.4-alpha3 without bload
* The culling is more pronounced with higher bloat
* I had the feeling that there is kind of a ceiling at where the culling did not become worse (e.g. 1000k was not noticably worse than 500k) but I also don't have a proper test methodology right now

m-m-adams, 2023-08-16:
I am getting much better clone performance with the move to external allocation 

https://github.com/SynthstromAudible/DelugeFirmware/assets/52469396/3dda10c5-1511-4198-92c4-549de1d87750

PaulFreund, 2023-08-19:
[CopyCullingExample.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/12385581/CopyCullingExample.zip)

Unfortunately the culling on copy is still very noticable. I created a very simple test song including my simple ACID preset and a stock drum kit collected in a folder. Just open the song in rows song layout and copy the drum track multiple times while playing to reproduce. 

Even if this is not neccessarily from the memory limitations we should fix it by feeding the audio engine druing copy / check why it takes so long. (probably reallocation/memmove?)

PaulFreund, 2023-08-31:
It seems this issue is really starting once the internal memory is no longer able to accomodate objects that are related to voice rendering such as Voice class objects. To test src/deluge/memory/general_memory_allocator.cpp can be modified as following:

```C++
void* GeneralMemoryAllocator::alloc(uint32_t requiredSize, uint32_t* getAllocatedSize, bool mayDeleteFirstUndoAction,
                                    bool mayUseOnChipRam, bool makeStealable, void* thingNotToStealFrom,
                                    bool getBiggestAllocationPossible) {

	if (lock) {
		return NULL; // Prevent any weird loops in freeSomeStealableMemory(), which mostly would only be bad cos they could extend the stack an unspecified amount
	}

	if (mayUseOnChipRam
#if TEST_GENERAL_MEMORY_ALLOCATION
	    && getRandom255() < 128
#endif
	) {
		lock = true;
		//uint16_t startTime = *TCNT[TIMER_SYSTEM_FAST];
		void* address = regions[MEMORY_REGION_INTERNAL].alloc(requiredSize, getAllocatedSize, makeStealable,
		                                                      thingNotToStealFrom, getBiggestAllocationPossible);
		//uint16_t endTime = *TCNT[TIMER_SYSTEM_FAST];
		lock = false;
		if (address) {
			return address;
		}
		else {
			display->displayPopup("FULL Internal");
		}
	}
```

To get popups in that event to see the connection between crackling and internal memory being full

PaulFreund, 2023-09-22:
Fixed by #557

PR #358 (2023-08-17): Operator New Alloc to external only
Changes the new operator to allocate in external RAM. This frees a significant amount of internal RAM and regains some of our lost performance 

Note that with this change any performance critical objects must be manually allocated and initialized since new would place them on slower SDRAM. Luckily the audio related code already does this.

I believe this fixes  #350
closed 2023-09-22 by PaulFreund