SynthstromAudible/DelugeFirmware #367: Frequent audio dropouts even with songs with few tracks

soymonitus, 2023-08-18:
Almost 80% of my songs that I had saved on my Deluge, which played fine with official firmware 4.1.3, now with the build compiled for release from current community branch, are having lots of audio drop outs. From which point in the commit history of the community release did the CPU and RAM started being pushed to the limit? Is there any CPU intensive feature I can remove completely to create a build that at lease doesn't have drop outs? Maybe some setting I have to apply to build script apart from compiling for Release? Any plans on improving performance?

m-m-adams, 2023-08-19:
7SEG or OLED, and is that with a brand new build? Performance at the current head is significantly better than it was a week ago.

soymonitus, 2023-08-19:
> 7SEG or OLED, and is that with a brand new build? Performance at the current head is significantly better than it was a week ago.

Oled

soymonitus, 2023-08-19:
Well, it is somehow fixed. I was wrongly selecting the wrong build to flash to the Deluge. With the Release build it is working quite good. I still get some audio drops in a few songs but can fix it by using the Delete Unused kit rows feature (i was getting dropouts specially on songs with kits loaded with a big amount of unused samples. This seems to indicate that there is more RAM being used by the new features added by the community but can be solved by reducing the number of samples loaded at the same time. I feel it is a nice compromise now between a lot of super cool new features and having to take a little bit more of care planning your tracks

closed 2023-08-19 by soymonitus