SynthstromAudible/DelugeFirmware #371: A song with just and audio clip, when loaded, has the clip button flashing

soymonitus, 2023-08-20:
@seangoodvibes found this accidentally but I think it's related to the Automation feature

seangoodvibes, 2023-08-20:
Hrm it is possible this was my mistake. Can you confirm it doesn't flash with normal firmware?

Good catch

soymonitus, 2023-08-20:
I think the deluge thinks wrongly that it is inside automation clip view. There must be a code path that leads to this behavior for audio clips

seangoodvibes, 2023-08-20:
> I think the deluge thinks wrongly that it is inside automation clip view. There must be a code path that leads to this behavior for audio clips

Yea you're right there's a bug. Adding it to my list of things to fix.

Thanks!

seangoodvibes, 2023-08-24:
@soymonitus This has been fixed. Can you test and close the issue if you are satisfied :)? Thx

closed 2023-08-24 by soymonitus