SynthstromAudible/DelugeFirmware #375: Fill notes feedback

trappar, 2023-08-20:
Going to leave my feedback here since [the PR](https://github.com/SynthstromAudible/DelugeFirmware/pull/211) already been merged.

Overall I found this feature initially confusing. I wasn't able to understand how it worked by looking at the community features. I had to read through the PR to try understand it. It feels like maybe some of this can be chalked up to bugs since the community features explanation seems to conflict with what I'm seeing on my OLED. For example, it mentions "The FILL setting is at the start of the probability range, before 5%.", but I see no such setting.

### Feedback
* The fact that you have to set a note to "PROBABILITY: 0%" in order for it to be a fill note is not intuitive. Why not have it say something like "PROBABILITY: 0% FILL ONLY". 
* What does the "PROBABILITY: 0% LATCHING" option do?
  * Was that always there?
  * I can't seem to find any mention of this anywhere.
  * It only seems to exist when there are multiple different samples playing in a single step, why?
  * It doesn't seem to influence behavior at all. These notes seem to always play.
* Is there any faster way to mark a note as a fill note than hold the note and scroll way left on the select encoder? This seems like it requires too much interaction on a per note basis, though it's not totally out of line with the existing level of interaction required for probability I suppose.
* No visual feedback to indicate which notes are fill notes makes patterns difficult to understand.
  * Could have them half-illuminated, but maybe we should reserve that sort of thing for something more important?
  * Could have these notes flash when holding the fill trigger?
* On OLED in the Sync Scaling Action community features menu it should say "SCALE" instead of "SCAL"

soymonitus, 2023-08-21:
I will add that "Note row probability" new feature reaches only till 5% probability and it would be awesome if FILL feature is also implemented for full rows. I can visualize myself cloning a row with the same sample as another one and setting Probability:0% to the whole row so i know that the snares i put in the first row are "normal" snares amd if i place any not on the new row they will be "fill" snares. Much quicker to work like this. Pleas @weavermedia can you address this?

soymonitus, 2023-08-21:
Nevermind @weavermedia it was a one-liner fix
https://github.com/SynthstromAudible/DelugeFirmware/pull/384

m-m-adams, 2023-08-21:
Yeah it's just a timing thing, when I did note row probability fill didn't exist and vice versa

Changing the label on 0% to fill makese sense to me

m-m-adams, 2023-08-21:
> What does the "PROBABILITY: 0% LATCHING" option do?
>
>     * Was that always there?
>
>     * I can't seem to find any mention of this anywhere.
>
>     * It only seems to exist when there are multiple [...]

This is an inconsistency with the existing UI - latching basically 'rolls the dice' once when the clip starts, and all notes set to latching will use that value instead of re rolling every time there's a note on. It doesn't apply to fill so shouldn't be possible to set a latching value

trappar, 2023-08-21:
@weavermedia Just tried the latest build and all of my feedback stands.

weavermedia, 2023-08-21:
@trappar

 **"The FILL setting is at the start of the probability range, before 5%.", but I see no such setting.**
I assume this is because you're seeing "PROBABILITY: 0%"? Which needs to be changed to FILL on OLED. 

**The fact that you have to set a note to "PROBABILITY: 0%" in order for it to be a fill note is not intuitive**
I don't have an OLED so this is new to me. I'm guessing you should be seeing "PROBABILITY: FILL". This is likely a simple HAVE_OLED fix.

**What does the "PROBABILITY: 0% LATCHING" option do?**
Latching has always been there. But it does need to the removed from the fill feature. I had wondered about repurposing the latching fill to a NOT fill but I think the NOT feature needs to be a whole separate PR.

**Is there any faster way to mark a note as a fill note than hold the note and scroll way left on the select encoder?**
The FILL setting could be nestled between either the probability range and 100% or between the iteration range and 100% but I'm not the person to tackle that. The cleanest way to implement a fill setting was to overtake the zero value of the range. Moving the fill value to another place in the range will touch a lot of code and I'm not sure it's worth it. Something has to be at the end of each range, and someone will always want it to be something different. Setting a fill value is currently only one more click than 5% and several fewer clicks than 8of8.

**No visual feedback to indicate which notes are fill notes**
This is the same for all probability and iteration notes. All notes with probability or iteration conditions show normally on the display and will play or not play based on those conditions. Fill notes feel no different in this respect.

**On OLED in the Sync Scaling Action community features menu it should say "SCALE" instead of "SCAL"**
I can submit a PR to fix that, or feel free to jump in if you like.

trappar, 2023-08-23:
Left a bit of feedback in the PR.

trappar, 2023-08-24:
Now there's a way more serious issue:

* When multiple notes are in the the same row, the probability selection will start with `FILL`, `FILL`, `5%`, `10%`, ...etc. The first of these "FILL" options works as expected, but the second one is actually FILL LATCHING and doesn't seem to work at all. Notes with this second fill option set seem to play 100% of the time regardless of if fill mode is being triggered or not.

And these are still unaddressed:

* Is there any faster way to mark a note as a fill note than hold the note and scroll way left on the select encoder? This seems like it requires too much interaction on a per note basis, though it's not totally out of line with the existing level of interaction required for probability I suppose.
* No visual feedback to indicate which notes are fill notes makes patterns difficult to understand.

weavermedia, 2023-08-24:
@trappar I'll look at the fill latching issue when I have time (not for at least a week or two). The issue you described sounds about right. Any notes set to a latching value will sound if any other notes in the clip are set to the same non-latching value. So if the latching logic follows as normal, the latching fill notes will sound if there's another non-latching fill note in the clip, regardless of whether the fill mode is active or not.

As I said, I don't have time to dig into how the latching prevBase logic works for a while.

I've already addressed the last two points [here](https://github.com/SynthstromAudible/DelugeFirmware/issues/375#issuecomment-1686845681) and won't be doing any  work on them for those reasons. If you or someone else wants to move the fill value to somewhere else in the probability/iteration range I'm fine with that but it looks like it will touch a lot of code for not very much gain.

PR #450 (2023-08-26): Bugfixes for fill mode
Fixes #442 
Fixes #451 
Fixes #375
Fixes #385 

Fill notes can no longer have a latching value (it didn't do anything)
Fill notes can now be quickly set by holding fill and adding notes
New notes inherit song fill status or row probability by default 
Fill mode can be properly deactivated over midi CC as well as note on/off
closed 2023-09-01 by m-m-adams