SynthstromAudible/DelugeFirmware #378: Wave Folding param causes Error in Audio Clip

seangoodvibes, 2023-08-21:
@m-m-adams 

I discovered a bug.

1. Create audio clip
2. Press Shift + Fold grid shortcut (pad to the left LPF frequency)
3. Deluge freezes with error "E411"

m-m-adams, 2023-08-21:
I think this is related to the unpatched param stuff you told me about on Discord. The audio version of the menu should link to a seperate unpatched wavefold amount instead of the patched param

seangoodvibes, 2023-08-21:
> I think this is related to the unpatched param stuff you told me about on Discord. The audio version of the menu should link to a seperate unpatched wavefold amount instead of the patched param

Makes sense! I had a feeling it was related to that discussion also.

PR #448 (2023-08-25): Bugfix - filter and wavefold menu/shortcut fixes
Fix #378 by removing it

Flash soon for #391 (needs bigger changes and new unpatched parameters)

Allow HPF mode selection and filter route in audio clips since they already work

Change filter mode menus and popups to use l10n
closed 2023-08-25 by m-m-adams