SynthstromAudible/DelugeFirmware #402: Selecting a clip in Grid mode deactivates quantized stutter

suivaht, 2023-08-22:
In the latest build that includes quantized stutter (although it doesn't seem like this includes the automation pop-up fixes that were recently merged) I noticed that when I enabled quantized stutter in the runtime settings, went to the grid mode, turned on Affect All an then engaged stutter, if I keep holding the knob when I select a clip on the grid the stutter effect immediately stops.  This behavior does not happen with the row layout (stutter stays engaged when selecting clips or sections).

soymonitus, 2023-08-22:
I think this is a bug related to Grid mode feature, because when you engage a clip to play, the developer can decide if aborting the stutter going on or not.

PaulFreund, 2023-08-28:
Fixed in current release branch and nightly :)

commit 5bbd328d (2023-08-27): Fix #402 global stutter being disengaged on clip launching
closed 2023-08-28 by PaulFreund