SynthstromAudible/DelugeFirmware #412: Fill iteration mode cannot be set for notes which have been configured with Note Repeat

soymonitus, 2023-08-22:
When you hold a note and press and turn the up/down knob you increase the number of Note repeats of that note. If you also hold it and turn the Select knob you change the iteration mode for all the repetitions of that step. The bug is that the probability iterations only reach 5%, they don't go down till the Fill condition
@weavermedia

weavermedia, 2023-08-24:
Sorry, I don't have time to look at this right now. Happy for anyone else to jump in.

m-m-adams, 2023-08-26:
Can you check if this is now fixed? I wasn't specifically looking at it but might have gotten it anyway

soymonitus, 2023-08-29:
@m-m-adams this is fixed here:

https://github.com/SynthstromAudible/DelugeFirmware/pull/476

closed 2023-08-30 by m-m-adams