SynthstromAudible/DelugeFirmware #413: Turning < > while holding clip pad on grid view bug

mcabrerapf, 2023-08-22:
When in grid mode, while holding a clip pad and turning the < > encoder, the view will remain stuck in a semi clip view where you can move the gold encoders.

m-m-adams, 2023-08-22:
Can you confirm exactly what version you built? This was broken in the release branch from about midnight last night to noon today but should be fixed now. The community branch and nightly release shouldn't have had the issue

mcabrerapf, 2023-08-23:
Im on nightly version `deluge-v1_0_0-nightly+2023_08_23-af2538b`
Heres a video showing the issue

https://github.com/SynthstromAudible/DelugeFirmware/assets/21365117/ba8a3cde-30d4-477f-a917-4ae0d0937b76

seangoodvibes, 2023-08-23:
@PaulFreund FYI

PaulFreund, 2023-08-28:
Fixed in current release branch and nightly :)

commit ba790fb1 (2023-08-27): Fix #413 scroll not invalidating grid selection
closed 2023-08-28 by PaulFreund