SynthstromAudible/DelugeFirmware #427: MIDI Takeover Pickup hanging

PaulFreund, 2023-08-24:
**Describe the Issue**
If Pickup is selected for MIDI takeover mode the values from my external encoder is only picked up if changed very slowly and otherwise hangs. 
Might originate in #170

**Expected Behavior**

I expect the value to not change until the current value is met and then always change

**Deluge Version**
- [ ] 7SEG
- [x] OLED

**Firmware Version**

- [ ] Nightly
- [ ] Community
- [ ] Built from community
- [x] Built from release/1.0

_Firmware Version in the file name or display:_ d3171ebd549abb220505a5f74cf5235658075743

_Steps to Reproduce the Issue_

1. Enable MIDI takeover mode in MIDI settings
2. Open Clip and open value like Pan or OSC1 PW
3. Hold learn and turn knob
4. Turn the knob very fast over the current value and see that it only changes once
5. Turn very slowly to see it change continusoly

PaulFreund, 2023-08-24:
@seangoodvibes FYI

soymonitus, 2023-08-29:
This was fixed yesterday in referenced PR. Please close

closed 2023-09-01 by m-m-adams