SynthstromAudible/DelugeFirmware #436: New sync modes for lfos/delay/arp show up separated instead of being linear

mcabrerapf, 2023-08-24:
### Please describe the problem:

Currently new sync modes (triplets and dotted) show up after scrolling all values of a group (straight then triplet then dotted).

So for example to reach 8th dotted from 8th straight you would have to scroll half of all straight values + all triplet values + half of all dotted values which is a big jump.

### What is the expected behavior?

I think they should flow in either descending absolute values (8th dotted  - 8th - 8th triplet) or descending grouped values (8th - 8th dotted - 8th triplet)

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_08_23-af2538b

m-m-adams, 2023-10-14:
Closing but open to PRs - changing this without breaking compatibility with older presets is not trivial

closed 2023-10-14 by m-m-adams

ok-reza, 2023-10-17:
if a PR isn't made/unlikely that makes sync options linear as @mcabrerapf suggests without breaking compatibility, perhaps holding SHIFT while scrolling in order to keep the sync number the same, but adjust the division (eg, 16th notes is displayed, hold SHIFT, scroll, 16th-triplet is next, 16th-dotted after, scrolling further has no effect. if you release shift at 16th-dotted and scroll you would then see 32nd-dotted).