SynthstromAudible/DelugeFirmware #437: Row step change on keyboard view scrolls the layout when reducing row step below 1

mcabrerapf, 2023-08-24:
### Please describe the problem:

On a synth clip, with keyboard view on, if you reduce the Row Step after it has reached 1 it will scroll the keyboard view to the left (which is what should happen if you use the < > knob without shift).

### What is the expected behavior?

If you are changing the Row Step on keyboard view (shift + < >) it should never horizontally scroll keyboard view (regular < > behaviour) even if it has reached min or max values (1 - 16)

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_08_23-af2538b

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on deluge
2. It will default to a synth clip view
3. Press keyboard button
4. Change Row Step (Shift + < >) to the left AFTER it has reached value 1

This will scroll the keyboard view horizontally to the left after it has reached 1

m-m-adams, 2023-08-25:
Oddly I can't reproduce this on 7seg

m-m-adams, 2023-08-28:
Confirmed this happens on OLED with current release head

PaulFreund, 2023-08-28:
Copied code and accidentially masked another variable called offset. I think on 7Seg the displayed step number might have been wrong the whole time

soymonitus, 2023-08-29:
This is fixed in latest HEAD. Please close it

PR #464 (2023-08-28): Fixes #437 - Scrolling while changing row stepping in keyboard view
Fixes #437, copied code masked variable also named offset
commit dbaa8420 (2023-08-28): Fixes #437 (#464)
closed 2023-08-29 by m-m-adams