SynthstromAudible/DelugeFirmware #439: Automation View - Conflicting display information on MIDI tracks

trappar, 2023-08-25:
### Please describe the problem:

I'm not sure how to describe this without just putting in instructions on what I'm doing and the weird things I'm seeing so...

1. Boot up
2. Press MIDI
3. Press Clip - you should now be in automation view. Display says "MIDI CHANNEL 1"
4. Press the lower left most pad - display says "MIDI CHANNEL" in the background with popup "CC 0" on top. Should it be like this? Seems weirdly overlayed.
5. Press any grid pad - display briefly says "<VALUE>" in a popup with "MIDI CHANNEL 1" in the background, then switches to saying "CC 0 (AUTOMATED)" and no longer says "MIDI Channel" - maybe it's hidden behind the popup?

**Observation: When pressing a grid pad, it shows the value with "MIDI CHANNEL 1" in the background, making it appear that it has switched to the midi channel of whatever value you input, rather than affecting the CC's value.**

### What is the expected behavior?

I'm not sure, but it seems like popups make the functionality much more confusing in this case. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_08_24-d114fd1.bin

m-m-adams, 2023-08-28:
Yeah this definitely looks buggy - I think we need to move away from popups in the automation view

stellar-aria, 2023-10-03:
Closing as resolved via #543

closed 2023-10-03 by stellar-aria