SynthstromAudible/DelugeFirmware #454: Inconsistencies in Community Features Menu / Runtime Settings

seangoodvibes, 2023-08-26:
### Please describe the problem:

I noticed some inconsistencies in community features menu / runtime settings that I think should be fixed:

1. CatchNotes is shown as one word in the menu and should be two: Catch Notes

2. ModDepthDecimals should be formatted modDepthDecimals in the XML file to be consistent with the other runtime settings (first word starts lower case)

3. The community features menu requires short strings for 7SEG display

### What is the expected behavior?

See above

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

melokit, 2023-08-26:
last nightly firmware from the community firmware of today and a lot of 3notes3 disappears on my old track...

m-m-adams, 2023-08-26:
> last nightly firmware from the community firmware of today and a lot of notes disappears on my old track... 
> 
> on 7 seg

Could you open an issue and explain what you mean in a bit more detail?

stellar-aria, 2023-09-17:
Closing as fixed by #521

closed 2023-09-17 by stellar-aria