SynthstromAudible/DelugeFirmware #462: Tracks missing contents in old songs

Icoustic, 2023-08-28:
### Please describe the problem:

After loading a couple of old songs and playing back i.e. streaming audio files from SD, upon loading another, screen says "SD card error" then the song loads with tracks present but missing contents i.e. no files or colors in clips (when attempting to replicate, problem does not seem to arise when several old songs are loaded without actually playing inbetween). These are large songs with plenty of tracks and samples used.
Not having same issue with previous non-community official firmware, on same SD card.

Relevant info: Big SD card, 400GB

I can provide collect media save(s) for some songs if needed

### What is the expected behavior?

Old songs should load like they're supposed to

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

d56dfaa

### If possible provide the steps to reproduce the issue and upload additional media:

1. Power on 
2. Load song 1 in given setlist, play a few secs 
3 Load song 2 in setlist, play a few secs
4. Load song 3 in setlist, screen says "SD card error", song loads with missing contents

m-m-adams, 2023-08-28:
I can't reproduce this - it would be great if you could share some of the songs with issues

Icoustic, 2023-08-29:
> I can't reproduce this - it would be great if you could share some of the songs with issues

Sure, here's a wetransfer link to 3 songs which should be enough to replicate the issue. The issue can happen with any of the songs, depending on what order they are opened and played in.

I'm guessing this is some memory leak thing becoming apparent due to the songs being big and packed.

https://we.tl/t-h0jjB4pATW

m-m-adams, 2023-09-01:
Likely related to #350 and #491

Icoustic, 2023-09-02:
> Likely related to #350 and #491

Seems like it!

PaulFreund, 2023-09-22:
Can you please retest with tomorrows nightly (probably generated in ~2h)?

Icoustic, 2023-09-22:
> Can you please retest with tomorrows nightly (probably generated in ~2h)?

Sure 🙂

Icoustic, 2023-09-23:
> Can you please retest with tomorrows nightly (probably generated in ~2h)?

Yep, its working now. The track I linked in #491 plays like it should/does on the official firmware, and I’m no longer getting the SD card errors 👍

Icoustic, 2023-09-23:
Ok hold on I am still getting the SD card errors but it takes longer i.e. loading and playing more songs. Before, I’d consistently get it after loading 3 songs, but now its like 7.

PaulFreund, 2023-09-23:
Okay it seems like your error is dependent on how much memory is available, that is a good clue! Can you try testing the same with official and push it with loading more?

Icoustic, 2023-09-23:
> Okay it seems like your error is dependent on how much memory is available, that is a good clue! Can you try testing the same with official and push it with loading more?

Yeah, did that, SD errors dont happen with official it seems. I’m guessing it clears out memory somehow between songs; on nightly, the error only becomes apparent with large and sample heavy songs, particularly with many audio clips, and I presume it stacks up and overloads

Icoustic, 2023-10-13:
> > Can you please retest with tomorrows nightly (probably generated in ~2h)?
> 
> Yep, its working now. The track I linked in #491 plays like it should/does on the official firmware, and I’m no longer getting the SD card errors 👍

Dropout issues are back, audio clip stops playing mid-sequence, and all audio clips are subsequently broken (won’t play all the way thru even though they did previously before the longest loop dropped out) and don’t return to normal until reboot

m-m-adams, 2023-10-14:
What firmware version string?

Icoustic, 2023-10-14:
> What firmware version string?

The nightly from the 9th that remained for a few days

m-m-adams, 2023-10-14:
yeah try again now, there was extra logging enabled that impacted performance

Icoustic, 2023-10-17:
> Ok hold on I am still getting the SD card errors but it takes longer i.e. loading and playing more songs. Before, I’d consistently get it after loading 3 songs, but now its like 7.

@PaulFreund FYI, this issue has gotten worse again, current beta. Happens after 3ish songs now (SD card error during loading, audio clip contents missing)

closed 2023-10-25 by m-m-adams

Icoustic, 2023-10-28:
As mentioned in #491 , still seeing these issues @m-m-adams

m-m-adams, 2023-10-29:
The current memory issues are being tracked in #643, we're aware of the underlying issue just not sure how best to fix it

PR #638 (2023-10-24): Memory management improvements
Join venture with @m-m-adams :D

* Reorganizes all GMA allocations or comments them if a possible change was not done
* Separates Stealable memory (56Mb) into it's own region
* Creates an 8Mb external region for all slower non stealable memory
* Moves FX, Delay and Grain FX buffer to extenral memory (only on Grain Fx this has a  minor negative performance impact)
* ~~Moves all parameters to external memory without negative impact massively increasing the amount of clips that can be created~~ This seems to make the UI sluggish and leads to culling on copying kits
 [... cut]