SynthstromAudible/DelugeFirmware #468: Stutter rate quantize

Rudement, 2023-08-29:
### Please describe the problem:

Clicking stutter rate quantize ON does not prompt for timing. 

### What is the expected behavior?

To give a choice 1/8,1/16, etc. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

1.0.0-NIGHTLY-D56DFAA

Rudement, 2023-08-29:
New to GitHub, I closed this accidentally.

bobtwinkles, 2023-08-29:
Hi, the Quantized Stutter feature is disabled by default because it changes existing behavior. You'll need to first enable it in the Community Features menu.

If you have it enabled, the stutter effect should be quantized, though it won't display popups. Instead, each LED next to the knob will either be full-on or full-off representing the different quantize levels (4ths/8ths/16ths/32nds/64ths respectively). We could use the OLED to display the stutter rate as well, but that's not implemented today.

m-m-adams, 2023-08-29:
I believe this is working as intended - feel free to open a discussion instead if you think it could be improved but I'm closing this as it's not a bug

closed 2023-08-29 by m-m-adams