SynthstromAudible/DelugeFirmware #471: Saving settings to SD card can silently fail if card is in RO mode

bobtwinkles, 2023-08-29:
### Please describe the problem:

When writing out settings (MIDI device config, community features) the save will silently fail if the SD card is in RO mode.

### What is the expected behavior?

A warning should be presented indicating that saving settings didn't work. 

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

nightly, community, Built from community, Built from release/1.0

### What is the firmware name:

dbaa84209d7edf7079366871e8933a98635805d1

M0R172, 2024-01-31:
This should be closed, I guess.

sapphire-arches, 2024-01-31:
Fixed by #1001

closed 2024-01-31 by sapphire-arches